tangalbert919/gigabyte-laptop-wmi #1: setting threshold

maniacx, 2023-04-29:
Not an issue, but just need some info.
I would like to add gigabyte support on my gnome extension [Battery Health Charging](https://extensions.gnome.org/extension/5724/battery-health-charging/) for setting charging threshold. (I dont have gigabyte laptop to test)

1. Is this kernel module ready and working?

2. How do I set charging threshold? after reading it seems way to set threshold using 2 commands.
`echo 1 | sudo tee /sys/devices/platform/gigabyte-laptop/charge_mode` 
`echo 60 | sudo tee /sys/devices/platform/gigabyte-laptop/charge_limit`
Charging Limit can be in the range of 60-100.
Is the path correct? 

3. Can the charging-limit be read when charging-mode is set to zero?
Fox example if I do set threshold to 80 and later disable mode and read value of charge-limit in this sequence of command

`echo 1 | sudo tee /sys/devices/platform/gigabyte-laptop/charge_mode` 
`echo 80 | sudo tee /sys/devices/platform/gigabyte-laptop/charge_limit`
`echo 0 | sudo tee /sys/devices/platform/gigabyte-laptop/charge_mode` 
`cat /sys/devices/platform/gigabyte-laptop/charge_limit`
What would be the result of the final **cat** command? would it be 80, 100 or would it error out?

tangalbert919, 2023-05-01:
1. For the most part, everything I needed working is working.
2. That is how you set the charging threshold, but replace `gigabyte-laptop` with `gigabyte_laptop`.
3. When charging mode is set to 0, the charging limit can still be read and modified. The result of your final **cat** command would be 80 in your example.
4. No difference there (at least from my observations).

Just as a disclaimer, I can only verify that my kernel driver works on Aero and AORUS models.

maniacx, 2023-05-01:
I have add gigabyte laptop, to my extension on Github, 

https://github.com/maniacx/Battery-Health-Charging

But I havent still not pulished this new version on gnome extensions yet, will do it in a week.
Thank you for the information.