tangalbert919/gigabyte-laptop-wmi #2: Can not install on Didabyte AERO 15 XD

Infraviored, 2023-07-28:
The Console dialog says it all.

flo@aero-ubuntu20:~/Programs/gigabyte-laptop-wmi$ sudo insmod gigabyte-laptop.ko
insmod: ERROR: could not insert module gigabyte-laptop.ko: Operation not permitted
flo@aero-ubuntu20:~/Programs/gigabyte-laptop-wmi$ mokutil --sb-state
SecureBoot disabled
Platform is in Setup Mode
flo@aero-ubuntu20:~/Programs/gigabyte-laptop-wmi$ modinfo gigabyte-laptop.ko
filename:       /home/flo/Programs/gigabyte-laptop-wmi/gigabyte-laptop.ko
version:        0.01
license:        GPL
description:    Gigabyte laptop WMI driver
author:         Albert Tang
srcversion:     6B3FF0BDDE41512A70DEF86
depends:        wmi
retpoline:      Y
name:           gigabyte_laptop
vermagic:       5.15.0-78-generic SMP mod_unload modversions 
flo@aero-ubuntu20:~/Programs/gigabyte-laptop-wmi$ dmesg | tail
[  410.634105] gigabyte_laptop: Cannot get handle
[  410.634108] gigabyte_laptop: Probe failed
[  868.219986] gigabyte_laptop: Cannot get handle
[  868.219990] gigabyte_laptop: Probe failed
flo@aero-ubuntu20:~/Programs/gigabyte-laptop-wmi$ lsmod | grep wmi
snd_rawmidi            49152  1 snd_seq_midi
asus_wmi               40960  0
sparse_keymap          16384  1 asus_wmi
platform_profile       16384  1 asus_wmi
wmi_bmof               16384  0
snd_seq_device         16384  3 snd_seq,snd_seq_midi,snd_rawmidi
snd                   102400  21 snd_hda_codec_generic,snd_seq,snd_seq_device,snd_hda_codec_hdmi,snd_hwdep,snd_hda_intel,snd_hda_codec,snd_hda_codec_realtek,snd_timer,snd_compress,snd_soc_core,snd_pcm,snd_rawmidi
wmi                    32768  2 asus_wmi,wmi_bmof
video                  61440  2 asus_wmi,i915

tangalbert919, 2023-08-07:
When you can, follow [this guide](https://wiki.ubuntu.com/Kernel/Reference/WMI) to get the WMI method GUIDs.

tangalbert919, 2024-01-26:
Try testing with the newer code. It should work on the XD model this time.

Infraviored, 2024-01-26:
> Try testing with the newer code. It should work on the XD model this time.

Going to test out in the coming days. Thanks a lot for updating me!

tangalbert919, 2025-01-15:
This issue is most likely fixed now, since WMI strings aren't used anymore. Closing.

closed 2025-01-15 by tangalbert919