tangalbert919/gigabyte-laptop-wmi #6: Adding GPU boost support

tangalbert919, 2024-03-22:
Starting with the [Aero 15 (RTX 20 Series)](https://www.gigabyte.com/Laptop/AERO-15--RTX-20-Series), Gigabyte's Control Center has an option for boosting NVIDIA GPUs. For the 2070 Max-Q, the "boost" just increases the GPU's TDP by just 10 Watts, from 80W to 90W.

The embedded controller is able to handle this, but how it's handled is inconsistent. From V9 to VB (and AORUS VC), this was the logic:
```
Case (0x51)
    {
        DBG8 = 0x51
        ^^PEG0.PEGP.TEMQ = Arg2
        ^^PEG0.PEGP.NLIM = One
        Notify (^^PEG0.PEGP, 0xC0) // Hardware-Specific
    }
```

However, from VC (and AORUS VD) onwards, this is the new logic:
```
Case (0x51)
    {
        Local0 = ToInteger (Arg2)
        If ((Local0 == Zero))
        {
            Debug = "<<< Opt-mode 2 disable >>>"
            ^^^NPCF.CTGP = Zero
            Notify (NPCF, 0xC0) // Hardware-Specific
        }

        If ((Local0 == One))
        {
            Debug = "<<< Opt-mode 2 Enable >>>"
            ^^^NPCF.CTGP = One
            Notify (NPCF, 0xC0) // Hardware-Specific
        }

        If ((Local0 == 0x02))
        {
            Debug = "<<< Qboost Diable >>>"
            ^^^NPCF.DTGP = One
            Notify (NPCF, 0xC0) // Hardware-Specific
        }

        If ((Local0 == 0x03))
        {
            Debug = "<<< Qboost Enable >>>"
            ^^^NPCF.DTGP = One
            Notify (NPCF, 0xC0) // Hardware-Specific
        }
    }
```
The models with the old logic did have a way to check if the boost is enabled through `WMBC`, but the models with the new logic do not. I'm not even sure what "Qboost" is, since I do not have any of the newer Aero models for testing.

morialo3, 2025-05-15:
I looked around find out what `Qboost` is, apparently `Qboost` and `Dynamic Boost` are the same.
`Qboost` from [Microsoft article](https://techcommunity.microsoft.com/blog/surfaceitpro/surface-laptop-studio-2-thermal-and-power-management-innovations/3949302):

> * CPU & dGPU power – When Surface Laptop Studio 2’s fans reach maximum speed, and the chassis temperatures must be held constant via CPU and dGPU power restrictions, we enable a unique solution that enhances NVIDIA's **QBOOST** technology. To optimize the user experience, we implemented runtime power steering controls to reallocate power between the CPU and dGPU, allowing for better performance at the same total power.

the article gives a hint and not a definition of what `Qboost` is.

From [Nvidia's Linux driver docs](https://us.download.nvidia.com/XFree86/Linux-x86_64/570.144/README/dynamicboost.html):

> The nvidia-powerd daemon provides support for the NVIDIA Dynamic Boost feature on Linux platforms. Dynamic Boost is a system-wide power controller which manages GPU and CPU power, according to the workload on the system. By shifting power between the GPU and the CPU, Dynamic Boost can deliver more power to the component that would benefit most from it, without impacting the system's total thermal and electrical budgets. This optimizes overall system performance per watt.

Here's a screenshot from Gigabyte's control center on my device:

[image]

You can see under `device information and power setup` knobs for cpu and gpu, that can be toggled on/off by the user. I hope this was helpful by any means.

morialo3, 2025-05-16:
Look like you already added the `gpu boost`, with latest release, looking at my gpu PL reading I see:
```shell
nvidia-smi -q -d POWER      

==============NVSMI LOG==============

Timestamp                                 : Fri May 16 14:54:10 2025
Driver Version                            : 570.144
CUDA Version                              : 12.8

Attached GPUs                             : 1
GPU 00000000:01:00.0
    GPU Power Readings
        Average Power Draw                : 30.39 W
        Instantaneous Power Draw          : 30.43 W
        Current Power Limit               : 105.00 W
        Requested Power Limit             : 105.00 W
        Default Power Limit               : 80.00 W
        Min Power Limit                   : 1.00 W
        Max Power Limit                   : 105.00 W
    Power Samples
        Duration                          : Not Found
        Number of Samples                 : Not Found
        Max                               : Not Found
        Min                               : Not Found
        Avg                               : Not Found
    GPU Memory Power Readings 
        Average Power Draw                : N/A
        Instantaneous Power Draw          : N/A
    Module Power Readings
        Average Power Draw                : N/A
        Instantaneous Power Draw          : N/A
        Current Power Limit               : N/A
        Requested Power Limit             : N/A
        Default Power Limit               : N/A
        Min Power Limit                   : N/A
        Max Power Limit                   : N/A
```
which used to stuck to only 90 W before update to `0.1.0-rc1`, well done 🎉

tangalbert919, 2025-05-17:
Getting GPU boost to work wasn't the issue I had. It was just figuring out if it was already enabled on boot, and newer models seem to lack any way of checking that.

I'll just close this because it doesn't look like any of this gets saved into EC memory anyway, so assuming that it is disabled on boot is my only option.

closed 2025-05-17 by tangalbert919

morialo3, 2025-05-17:
> I'll just close this because it doesn't look like any of this gets saved into EC memory anyway, so assuming that it is disabled on boot is my only option.

From my `dsdt`, looks like it searches register `0xE1` for a value:

```
Case (0x51)
                {
                    DBG8 = 0xE1
                    Local0 = ToInteger (Arg2)
                    If ((Local0 == Zero))
                    {
                        Debug = "<<< Opt-mode 2 disable >>>"
                        ^^^NPCF.CTGP = Zero
                        Notify (NPCF (), 0xC0) // Hardware-Specific
                    }

                    If ((Local0 == One))
                    {
                        Debug = "<<< Opt-mode 2 Enable >>>"
                        ^^^NPCF.CTGP = One
                        Notify (NPCF (), 0xC0) // Hardware-Specific
                    }

                    If ((Local0 == 0x02))
                    {
                        Debug = "<<< Qboost Diable >>>"
                        ^^^NPCF.DTGP = One
                        Notify (NPCF (), 0xC0) // Hardware-Specific
                    }

                    If ((Local0 == 0x03))
                    {
                        Debug = "<<< Qboost Enable >>>"
                        ^^^NPCF.DTGP = One
                        Notify (NPCF (), 0xC0) // Hardware-Specific
                    }
                }
```
`DBG8` is mentioned earlier as:

```
OperationRegion (DEB0, SystemIO, 0x80, One)
        Field (DEB0, ByteAcc, NoLock, Preserve)
        {
            DBG8,   8
        }
```

Are my assumptions right?, disclaimer: I'm not an expert, it's just a hoppy.