tangalbert919/gigabyte-laptop-wmi #9: Working until reboot

TipJar420, 2024-06-26:
Could someone help with: I just got Fedora Working well on my Aorus Lappy. I was trying to use extensions .gnome.org for my battery to only charge 75%. It seems to work until ReBoot. I know I had to install some dependancies  module but am not sure if I put it in the proper directory. I disapears every reboot,...

TipJar420, 2024-06-26:
I see now if I keep "sudo insmod gigabyte-laptop.ko" every reboot it works from the install dir. What is going on that it is not being read by the bootup sequence,....?

tangalbert919, 2024-09-10:
I'm guessing Fedora doesn't automatically load kernel modules unless they are installed with their package manager. They don't seem to like using DKMS, as they have their own "kmod" standard.

For now, you will just have to load the module manually, but I'll see if I can support that kmod standard. That way, you can reinstall the kernel module through the package manager and Fedora can load it automatically.

I've never used Fedora before, so this might take a while.

majidmadadi, 2024-09-12:
> I'm guessing Fedora doesn't automatically load kernel modules unless they are installed with their package manager. They don't seem to like using DKMS, as they have their own "kmod" standard.
>
> For [...]

I use also fedora. Which distro works perfectly with this?

tangalbert919, 2024-10-06:
The distro shouldn't matter. The kernel module should just work anyway.

tangalbert919, 2025-06-12:
Closing this, as this driver can now be loaded automatically on boot using DKMS.