tangalbert919/gigabyte-laptop-wmi #13: Doesn't work with AORUS 16X ASG

bouncytorch, 2024-10-02:
Mint 22, installed via dkms. 

```
bouncytorch@AORUS:~$ ls /sys/devices/platform
 ACPI0007:20   ACPI0007:29   ACPI0007:32   ACPI0007:3b    coretemp.0          kgdboc        PNP0C0B:04     rtc-efi.0
 ACPI0007:21   ACPI0007:2a   ACPI0007:33   ACPI0007:3c    efivars.0           microcode     PNP0C0C:00     serial8250
 ACPI0007:22   ACPI0007:2b   ACPI0007:34   ACPI0007:3d    eisa.0              MSFT0101:00   PNP0C0E:01     snd-soc-dummy
 ACPI0007:23   ACPI0007:2c   ACPI0007:35   ACPI0007:3e   'Fixed MDIO bus.0'   NVDA0820:00   PNP0C14:02     uevent
 ACPI0007:24   ACPI0007:2d   ACPI0007:36   ACPI0007:3f    i8042               pcspkr        PNP0C14:03
 ACPI0007:25   ACPI0007:2e   ACPI0007:37   ACPI0008:00    INT33A1:00          PNP0C0B:00    power
 ACPI0007:26   ACPI0007:2f   ACPI0007:38   ACPI000C:00    INTC1085:00         PNP0C0B:01    reg-dummy
 ACPI0007:27   ACPI0007:30   ACPI0007:39   ACPI000E:00    INTC10A0:00         PNP0C0B:02    regulatory.0
 ACPI0007:28   ACPI0007:31   ACPI0007:3a   acpi-cpufreq   intel_rapl_msr.0    PNP0C0B:03    rtc_cmos
```

tangalbert919, 2024-10-04:
You still have to load the kernel module, but looks like you found another problem here.

homuwell, 2025-01-23:
Hello there,

First of all, thank you for your work on this project. I also own an AORUS 16X ASG and have encountered some issues.
I cloned the repository and attempted to install it using `make -si`, but DKMS consistently fails and returns errors. To work around this, I had to build the package manually, add it to DKMS, and then it installed successfully. The module works perfectly after a reboot. However, I’m concerned about how it will behave during future kernel updates.
```
[homuwell@homuwell-arch ~]$ dkms status
aorus-laptop/0.0.3, 6.12.10-arch1-1, x86_64: installed
nvidia/565.77, 6.12.10-arch1-1, x86_64: installed
openrazer-driver/3.9.0, 6.12.10-arch1-1, x86_64: installed
[homuwell@homuwell-arch ~]$ ls /sys/devices/platform/gigabyte_laptop/
charge_limit      fan_mode   uevent
charge_mode       hwmon      usb_charge_s3_toggle
driver            modalias   usb_charge_s4_toggle
driver_override   power
fan_custom_speed  subsystem
```
The main issue, however, is that fan control is almost non-functional. Changing the fan mode does not seem to have any effect, except for "silent mode." Additionally, the custom fan speed option does not work when custom mode is enabled (whether in fixed or auto mode).
```
[homuwell@homuwell-arch ~]$ cat /sys/devices/platform/gigabyte_laptop/fan_mode 
5
[homuwell@homuwell-arch ~]$ cat /sys/devices/platform/gigabyte_laptop/fan_custom_speed 
100
[homuwell@homuwell-arch ~]$ sensors
...
gigabyte_laptop-isa-0000
Adapter: ISA adapter
fan1:        1871 RPM
fan2:        1833 RPM
fan3:           0 RPM
fan4:           0 RPM
temp1:        +56.0°C  
temp2:        +44.0°C  
temp3:        +56.0°C  
...
```
If additional information is needed, please let me know.
**System info:**
- OS: Arch Linux
- Kernel: 6.12.10-arch1-1

tangalbert919, 2025-03-28:
I've been debugging this on my AORUS 15 BKG for a while now, and I have noticed the issue you are having. This seems to be less about the driver and more about how Gigabyte defined the behaviors of their fan modes.

Can you see if 0.1.0-rc1 works for you? The sysfs path is renamed to `aorus_laptop` in that version, but it should fix some of the issues you're having on the 16X ASG.

homuwell, 2025-07-30:
> I've been debugging this on my AORUS 15 BKG for a while now, and I have noticed the issue you are having. This seems to be less about the driver and more about how Gigabyte defined the behaviors of [...]

After using your driver, I encountered several issues which I believe may be related to the embedded controller.

After switching back to Windows, the fan control in Gigabyte Control Center stopped working correctly. For example, selecting the Turbo mode no longer increases fan speed as it did previously.

On Linux, I also experienced a recurring issue where the GPU would randomly fall off the bus. This happened even after uninstalling your software. I assume this may have been caused by overheating due to improper fan control or other EC-related behavior. The problem was eventually resolved after a GCC update, which I believe also updated the embedded controller firmware.

I suspect that your driver might be modifying or interfering with the embedded controller firmware or its driver, which could explain the issues I encountered on both Windows and Linux.

Because of these issues, I am hesitant to test your driver again, as it might cause similar problems with the embedded controller. I would consider trying it again if you can provide clear instructions on how to fully restore the original Gigabyte firmware or drivers in case anything goes wrong.

bouncytorch, 2025-07-30:
> On Linux, I also experienced a recurring issue where the GPU would randomly fall off the bus. This happened even after uninstalling your software. I assume this may have been caused by overheating [...]

I am fairly certain that that problem is unrelated to this software in general, since I experienced it myself and "fixed" it.
For me, setting NVIDIA PRIME to Performance mode (dGPU) with the GPU driving the screen option in BIOS set as "Dynamic/MSHybrid" would guarantee the GPU will fall of the bus. The way to fix that would be to set the screen mode to Discrete in BIOS in case you want to use PRIME's dGPU mode. Also, it was never fixed for me so I have no idea what kind of update you got and how :)

I'm not remotely knowledgeable enough to speak on whether this is anything related to the embedded controller or not, I just *feel* like it isn't. Since I had a similar experience I decided it'd be worth sharing.

homuwell, 2025-08-03:
Getting back  with feedback @tangalbert919 . There are no issues with the installation anymore—everything is working correctly now (I just need to figure out how to write a Pacman hook so the module works after updates). The fan control modes are also fully functional and allow for easy switching between modes. Below is the output of all commands.

```
[homuwell@homuwell-arch aorus_laptop]$ cat /sys/devices/platform/aorus_laptop/fan_mode 
5
```

```
[homuwell@homuwell-arch aorus_laptop]$ cat /sys/devices/platform/aorus_laptop/fan_custom_speed 
70
```

```
[homuwell@homuwell-arch aorus_laptop]$ sensors
...
aorus_laptop-isa-000a
Adapter: ISA adapter
fan1:        4549 RPM
fan2:        4587 RPM
fan3:           0 RPM
fan4:           0 RPM
temp1:        +48.0°C  
temp2:        +37.0°C  
temp3:        +48.0°C  
...
```

I haven't done a stress test for stability yet, but I believe everything will be fine. Thanks again for your work. I think the issues can be closed now, and the AORUS 16X ASG can be added to the list of tested models. If anything goes wrong, I’ll open a new issue.

closed 2025-08-20 by tangalbert919