tangalbert919/gigabyte-laptop-wmi #14: Error inserting module

tiago1405, 2025-05-15:
Hello,

First of all, thank you very much for your work on what is a  super niche project but one of my favorite things I discovered since installing Linux on my '18 Aero 15xV8!

Previously I had been able to install with no problem but when I attempted to reinstall the drivers lately I have been getting the following error when inserting the module using `sudo insmod aorus-laptop.ko`:

```
insmod: ERROR: could not insert module aorus-laptop.ko: Invalid module format
```

I will also include below my sys info:
OS: Pop!_OS 22.04 LTS x86_64
Kernel: 6.12.10-76061203-generic
CPU: Intel i7-8750H (12) @ 4.100GHz
GPU: NVIDIA GeForce GTX 1070 Mobile

Please let me know if there's any more info I need to provide to you!

tangalbert919, 2025-05-17:
Did you update your kernel recently? If you did, you have to rebuild the module too.

tiago1405, 2025-05-18:
I may have updated the Kernel during one of the system updates. It does however seem to be the same Kernel as what the driver is for. And if by rebuild the module you mean run Make again, I have.
```
user@pop-os:~/Downloads/driver$ make
make -C /lib/modules/6.12.10-76061203-generic/build M=/home/user/Downloads/driver modules
make[1]: Entering directory '/usr/src/linux-headers-6.12.10-76061203-generic'
  CC [M]  /home/user/Downloads/driver/aorus-laptop.o
  MODPOST /home/user/Downloads/driver/Module.symvers
  CC [M]  /home/user/Downloads/driver/aorus-laptop.mod.o
  CC [M]  /home/user/Downloads/driver/.module-common.o
  LD [M]  /home/user/Downloads/driver/aorus-laptop.ko
make[1]: Leaving directory '/usr/src/linux-headers-6.12.10-76061203-generic'
user@pop-os:~/Downloads/driver$ sudo insmod aorus-laptop.ko
[sudo] password for user:
insmod: ERROR: could not insert module aorus-laptop.ko: Invalid module format
```

tangalbert919, 2025-05-28:
Do you have secure boot enabled? That might also be causing the problem.

closed 2025-06-09 by tiago1405