tangalbert919/gigabyte-laptop-wmi #17: Charging mode set to custom crashes Aero 15x

truvatech, 2025-07-11:
Hello, after almost 3 years of using Linux I can't believe I only now found this kernel module!
I have managed to add it to my Nixos build as a custom module and it loads properly on boot but there is only one thing...
Setting /sys/devices/platform/aorus_laptop/charge_mode to 1 seems to work, but not even a minute or two later my computer just... Shuts off without warning.

Is there any information I can provide you from my late 2018 Gigabyte Aero 15x (i7-8750H) that would allow us to get this working by any chance?

Thanks again

Dmatcha, 2025-09-09:
Also using an old 2018 Aeo 15x v8 (8750h) and cannot get this to work at all. Brightness keys, nothing works. Used yay and used the ldtarball method.

ymrtech, 2026-10-30:
well, not being able to use this caused my battery to expand and break out of my chassis LOL

I now no longer have a battery until i replace it but when I get a new battery it'd be nice to be able to set this to limit charging at 80% seeing as it's connected to power 24/7 usually

I have not tested this in months but will get back to you shortly to see if it still an issue. I don't know if i can test it without a battery though ha

jgoestl, 2026-11-12:
Toda I tried to set the charge mode on my Aero 15x and it chrashes seconds later.
Can I anything do to help investigating it?